PolySync/oscc #17: Estop STL file

droter, 2016-12-15:
I tried printing the estop stl file on a makerbot mini and it isn't able to print.  

Has anyone tried printing this file?

ltbj, 2016-12-15:
At what point does the print fail?  

Does the makerbot software generate the gcode from the STL properly?  

Any other info you can provide would be helpful! 

I'm not familiar with the makerbot mini, though it does have a fairly small print bed, only 4 x 5 inches. I'll take another look at the model and check the size.

droter, 2016-12-15:
It may be a problem with the maker bot.  It fails at the very beginning stalls on 77% after homing.  We are trying the export from solid works again.

There isn't any error messages.  I can test it again.

droter, 2016-12-20:
Must be something with the makerbot mini and orientation of the part.  We flipped it over so the switch holes were on top and it printed okay.

closed 2016-12-20 by droter