PolySync/oscc #21: Steering in higher speeds

zwn, 2016-12-30:
Can you please elaborate more on the steering implementation? AFAIU the power steering in most vehicles is active only in slow speeds. In higher speeds it disengages since it is not needed (and might even be counterproductive). Is the Kia Soul steering implementation different? 

In the [CAN Specification](https://github.com/PolySync/OSCC/wiki/CAN-Specification) I see a Steering Command - is the steering completely mechanical decoupled? I mean is it steer-by-wire? Or is there something else in the design I am missing?

evanlivingston, 2016-12-30:
@zwn The steering system in the Kia Soul is an electronic power steering system. The steering column has a DC motor, redundant torque sensors and a steering microprocessor. When a driver turns the steering wheel the torque in the column increases. The steering microprocessor then attempts to reduce the torque by powering the DC motor in the direction of the torque.

By placing the steering module between the torque sensors and the steering microprocessor, OSCC can cause the steering microprocessor to steer the vehicle. 

The CAN Steering Command messages are interpreted by the OSCC steering module, which translates the CAN messages into analog voltages then supplied to the steering microprocessor. The Steering Command messages are not sent to any Kia microprocessor, they are only available to the OSCC modules.

closed 2016-12-30 by evanlivingston

zwn, 2017-01-01:
Did you have a chance to test your implementation at speeds above 30km/h? I am wondering if the Kia steering microprocessor stops the assisting when a certain speed is reached as that is the behavior to expect when the car is driven by a human.

zwn, 2017-01-01:
I have not been able to confirm that the Kia contains a true steering-by-wire system. It seems to be only electric power assisted steering (EPAS) system. For example the page 
http://www.kisouman.com/electric_power_steering_eps_-465.html
suggests the motor might turn off to prevent its overheat and there might be other conditions as well. The loss of the power assistance is not treated as a dangerous situation which makes sense. It would be dangerous only in lane keep assist system.

I think the only car on the market with steer-by-wire is Infinity Q50. But I would love to be proved wrong :smile: