PolySync/oscc #23: Tests contain a lot of uncalled and commented-out code.

galv, 2016-12-31:
The test cases are currently really shabby. I'm not quite sure what they're meant to be doing right now. It seems almost as if they're supposed to be somehow interactive, but they're currently cluttered with commented out regions and unused functions.

Not sure how to resolve this issue. I just wanted to bring this up to the maintainers too understand what the goal is of the tests.

ltbj, 2017-01-04:
We can resolve this by cleaning up the existing code and adding useful comments -- changes will be pushed up soon and we'll come back to this issue with a PR.

ltbj, 2017-01-19:
This is addressed in the `bugfix/standardize-pid-library` branch, which will be pull-requested early next week once we tune the new code and validate the changes on the brake board. 

Validation on the steering board is complete, and the throttle board does not use the PID library. 

I'll leave this open until the PR gets through, and we have improved in-line code comments to descibe what tests should be doing, and where they are run.

closed 2017-01-31 by ltbj