PolySync/oscc #24: Duplication between `firmware/*/kia_soul_ps/*.ino` and `firmware/tests/*.ino`

galv, 2016-12-31:
There is a lot of duplicated constants between the actual firmware files and the test firmware files right now. I would really like to see these factored out into a common file that gets #included into these files instead. For example, `CAN_BAUD`, `SERIAL_BAUD`, and `CAN_INIT_RETRY_DELAY` get #defined everywhere to the same values. My experience is that you will probably never have any reason for these to be different between different boards (especially the serial and can bauds), anyway. And right now, as soon as you change something in the firmware, the tests could break, which is a headache.

I would like to see a common.h file that defines the above three constants, as well as a few others I'm not thinking of right now, and .h files for each type of board defining GPIO pin names, so we can rid ourselves of this duplication.

Also, I don't want to put multiple issues into the same github issue, but I realize now that the tests/ directory does not have the same directory structure as the rest of the code. i.e., there are no kia_soul_ps/ directories, even though the test there right now clearly (?) apply only to the kia soul.

lpdon, 2017-01-01:
Is there also any reason to have more than one PID code? OSCC/firmware/steering/kia_soul_ps/PID.* and OSCC/firmware/arduino_libraries/PID_v1/?

LucasBuckland, 2017-01-05:
Thanks for the input here guys. We have discussed this internally and will be adding a common.h file as well as standardizing the PID library etc in the near future.

evanlivingston, 2017-01-10:
@galv: Don't hesitate to create multiple issues.

The tests/ directory has slightly different structure because some tests are designed for the boards themselves rather than a car installation. The boards should have tests that can verify their core functionality separate from their functionality inside of a vehicle.

ltbj, 2017-01-19:
The `common.h` file was added and related changes were made in this PR: https://github.com/PolySync/OSCC/pull/38

galv, 2017-01-19:
@evanlivingston 

Is the intent that each PCB/arduino shield be able to interface to cars besides the Kia soul without having to refabricate the PCB? I haven't dug into their designs much, but was wondering about whether the team made that design decision.

ltbj, 2017-01-31:
You're right, the OSCC modules will work for other vehicles that have by-wire steering and throttle control modules that require voltage spoofing on redundant sensors.

We are currently looking at other vehicles that OSCC will be able to support, starting with other Kia models, specifically the Kia Niro.

closed 2017-01-31 by ltbj