PolySync/oscc #25: Create style guide.

galv, 2016-12-31:
I noticed a couple of commits that fix style in the repo, for example changing:

```
#define CAN_BAUD (CAN_500KBPS) -> #define CAN_BAUD ( CAN_500KBPS )
```

and 
```
if (foo) {

}
->
if (foo)
{

}
```

It would be super handy for contributors like me to know what the polysync style is so I can follow it from the get-go. Please add it to CONTRIBUTING.md or something like that.

galv, 2017-01-02:
I also noticed tabs mixed with spaces and dangling whitespace at the end of lines (emacs's whitespace modehighlights these issues for me). You should consider whether you want to require no tabs and to require that whitespace be removed at the end of lines (I think git can be set up via hooks to do this).

ltbj, 2017-01-04:
We'll get our style guide posted here shortly, thanks for the pointer! FYI we'll enforce spaces only, and no trialling whitespace.

evanlivingston, 2017-01-20:
@galv We have now added the coding standard! 

https://github.com/PolySync/OSCC/blob/devel/CONTRIBUTING.md

closed 2017-01-20 by evanlivingston