PolySync/oscc #28: Where to source parts for Brake installation

droter, 2017-01-05:
Hi,

I haven't been able to find plugs and brake lines to add the redundant brake.  Is there some place online where I can by the plugs and lines?

Thanks,

Matt

anathan, 2017-01-06:
Try Jegs or Summit Auto

-Aaron

droter, 2017-01-06:
Thanks Aarron,

Do you know the part size of the plugs?  Are they 10mm x 1.25 inverted?  Also, what are the lengths of the brake lines?

Matt

evanlivingston, 2017-01-06:
Matt,

We are working on creating a better BOM for these parts. The plugs for the brake actuator are -3. We use two 48" -3 brake lines and two 26" -3 brakes lines. 4 M12x1.0 inverted flare to -3 and 6 M10x1.0 inverted flare to -3 adapters are needed. After looking yesterday it looks like most of these parts are available from jegs and Summit Auto, except I haven't been able to find a reliable source of the adapters yet. If you find one please let us know so we can add it to the BOM.

droter, 2017-01-09:
Here are the parts I have found:

M10 1.0 inverted to -3 http://www.jegs.com/i/Russell/799/641431/10002/-1
48" Brake Line  http://www.jegs.com/i/Earls/361/63010148/10002/-1
26" Brake Line  http://www.jegs.com/i/Earls/361/63010126/10002/-1

Haven't been able to locate the M12 1.0 to 3 yet.

evanlivingston, 2017-01-10:
M12x1.0 to -3 Is a bit more rare. Supposedly Aeroquip makes them but I can't find a part number. Autobahn88 makes the part. Their part number is FTSS002-AN3-M121.

https://www.amazon.com/Autobahn88-Stainless-Steel-Fitting-Adapter/dp/B019RVZZ16

evanlivingston, 2017-01-10:
Okay, We have an updated and complete BOM for the brake install. Head over to https://github.com/PolySync/OSCC/wiki/Brake#hardware to check it out.

closed 2017-01-10 by evanlivingston