PolySync/oscc #32: Brake Module Box

droter, 2017-01-13:
Has anyone found a place to buy a brake module enclosure?

The 3d printed file will cost around $359 on shapeways.  Is there some other enclosure we can use?

Thanks,

Matt

evanlivingston, 2017-01-13:
@droter: 

The enclosure for the brake module currently in the repo is for the prototype board.

The new board is significantly smaller. I'm redesigning the enclosure today and doing test prints. The new enclosure should be cheaper to print but I agree an alternative should be found. We will spend some time here looking for project boxes that can accommodate the modules.

Evan

droter, 2017-01-18:
Would these type of boxes work:
http://www.graybar.com/store/en/gb/non-metallic-junction-box-93172343

evanlivingston, 2017-01-19:
Those work great. We used those for our prototypes. 

The board to control the brake actuator is 4"x8"x2.5"

closed 2017-01-20 by evanlivingston