PolySync/oscc #41: Structures are using bit fields

petebrink, 2017-01-20:
Bit fields are not necessarily safe and can vary according to what processor or compiler is used.  These should be changed to use bit macros (e.g. the Linux versions
#define SET_BIT(p,n) ((p) |= (1 << (n)))
#define CLR_BIT(p,n) ((p) &= (~(1) << (n))) )
to allow for the code to be more specific and safe.

petebrink, 2017-01-20:
Misplaced the parentheses on the CLR_BIT macro:
#define SET_BIT( bit, index )      ( ( bit ) |= 1 << ( index ) )
#define CLR_BIT( bit, index )      ( ( bit ) &= ~( 1 << ( index ) ) )
#define CHK_BIT( bit, index )      ( ( bit ) & ( 1 << ( index ) ) )

LucasBuckland, 2017-06-28:
This issue has been resolved in our recent code refactor.

closed 2017-06-28 by LucasBuckland