PolySync/oscc #81: [Enhancement] SocketCAN & CAN-Utils

bobjohnson2040, 2017-02-07:
A great enhancement would be to include CAN-Utils/SocketCAN for vehicle control. This would significantly lower the cost by replacing Kvaser Canlib and also increase compatibility with other CAN software/hardware.

CAN-Utils: https://github.com/linux-can/can-utils

SocketCAN: https://en.wikipedia.org/wiki/SocketCAN

ltbj, 2017-02-07:
We're familiar with SocketCAN, and adding support for OSCC is a great idea. I'll follow up with a timeline after talking with the PolySync team.

ltbj, 2017-03-21:
We've added an example of building applications against socketCAN in PR #114. 

You can see the joystick commander's `CMakeLists.txt` file to see how to compile the high level control node against the socketCAN libraries.

closed 2017-03-21 by ltbj