PolySync/oscc #84: Steering control should have a timeout for chassis_state_1 frame

evanlivingston, 2017-02-10:
The steering code depends on getting the steering angle from the C-CAN bus via the OBD-II connector. Currently, the firmware does not have a timeout for receiving the CAN frame that contains the steering angle. If the controller does not receive steering angle for any reason (disconnected OBD cable) then the steering controller runs but errantly.

petebrink, 2017-02-10:
This is going to required that we determine if there is any regularity to this message.  If the message does not get sent when the steering wheel is static, then it is impossible to determine if there has been a timeout.

Recommend doing some empirical testing to determine what the update rate for the chassis message is.

joshhartung, 2017-02-10:
IIRC they come at a 10ms update rate with very little jitter.

LucasBuckland, 2017-04-10:
This has been added to our backlog of tasks and will be addressed in the near future.

closed 2017-05-30 by austinjmorlan

austinjmorlan, 2017-06-01:
Resolved with #133.