PolySync/oscc #115: Throttle CS pins in code don't match schematics

hkorre, 2017-03-21:
Throttle code:
https://github.com/PolySync/OSCC/blob/devel/firmware/throttle/kia_soul_ps/throttle_control_module.ino#L49-L55
* #define CAN_CS                          ( 10 )
* #define DAC_CS                          ( 9 )

Schematic:
https://github.com/PolySync/OSCC/blob/devel/boards/sensor_interface_board/sensor_interface_board_sch_1.0.0.pdf
* Pin10 -> CS_A -> DAC_CS
* Pin11 -> CS_B -> CAN_CS

Seems to me that the schematic is incorrect.

dan-fern, 2017-03-21:
Hasan,

Thanks for pointing this out.  The diagram and firmware are both accurate in fact.  I think you mistook the pin number on the integrated circuit block to be the destination pin on the Arduino.  The Arduino pinouts are labeled on the inside of the `ARDUINO_JP1` block in zone C6 where `CS_A` runs to `D9` and `CS_B` runs to `D10`.  In Zone A3/A4, you'll see `CS_A` go to `MCP4922T`, which is the [DAC P/N](http://www.digikey.com/product-detail/en/microchip-technology/MCP4922T-E-SL/MCP4922T-E-SLCT-ND/736003).  In turn, `CS_B` goes to `MCP2515` in zone A6, which is the [CAN controller](http://www.microchip.com/wwwproducts/en/en010406).  I can see how it is confusing when the pin number and pin destination are offset by one.  

Best regards, 

Daniel

closed 2017-03-21 by dan-fern

hkorre, 2017-03-23:
@dan-fern: Thanks so much for clarifying!