PolySync/oscc #117: ROS Interface

bobjohnson2040, 2017-03-23:
Is there any future plans for a ROS interface? By adding it there would be more compatible open source software and it would encourage more robotics developers to join the OSCC movement!

LucasBuckland, 2017-04-10:
This is a great idea, thanks for your input! We will add this to our future tasks and I can update you when we end up implementing it.

andrewraharjo, 2017-04-13:
@bobjohnson2040 you can start leveraging the arduino code with `roslib`

bobjohnson2040, 2017-05-22:
Any update on this?

whitmanjack24, 2017-06-28:
+1 
I am also extremely interested in a ROS interface!! 

@LucasBuckland @k-cleary Are either of you panning to implement this?

LucasBuckland, 2017-06-28:
Hello, yes we are planning on a ROS interface and it is in our road-map. When I have more details I'll follow up on a time-line here. 
Thanks for the feedback!

whitmanjack24, 2017-06-30:
@LucasBuckland Thank you!

rebpdx, 2017-09-25:
A ROS node for OSCC is currently in progress. The node will allow for publish and subscribe to ROS messages based on the respective OSCC API types. It is planned to be released in mid October.

rebpdx, 2017-11-07:
The ROS node is now available, which can be cloned into a catkin workspace. https://github.com/PolySync/roscco

closed 2017-11-07 by rebpdx