PolySync/oscc #140: Supported Kia Soul models?

MohammadAbuQare, 2017-06-21:
Hi, 

Does OSCC support Kia Soul 2017?
What are the supported Kia Soul models (LX, EX, SX) ?
Does it need to have Auto Parking Assistant? 
Is there any modification or requirements needed?

MQ

jweatherson, 2017-06-22:
Hi MQ,

The kit does support the 2017 Kia Soul and the LX, EX and SX models all work with the kit.  It does NOT need to have Auto Parking Assist.  

Modifications to the brake lines are required when adding the Toyota brake controller.  In addition, without the factory connectors, it is required to splice into the steering column torque sensors and throttle position sensors.  

Thanks,
James

waiwnf, 2017-06-30:
Dear PolySync people, I am trying to extend your work (only wrt steering for now) to the European KIA Cee'd, but am having some issues. So I would appreciate if you could share some details on your setup:

1. What is the steering column part numbers you are working with? Is that 56310B2000 for the column and 56340B2000 for the MDPS ECU?

2. I have a 2012 Cee'd steering column with MDPS (part 56300A2000) with MDPS ECU connectors to the sensors and steering drive shown at this diagram http://kiaceed2.ru/shemy/el-us.png . Does that look similar to what you are working with?

Apparently the number of wires is different on the Cee'd (7 vs 6 for Soul). There wires for VTAS, TS4 (?), SAS1, SAS2, TSS, Ground and +5V. What puzzles me is that (at rest) from SAS1 comes +5V and from SAS2 comes 0V, which does not match your plot for torque sensor voltages. Also both TS4 and TSS produce 0.

3. Have you tried "booting up" power steering separately on a test bench without a car? I don't have the car itself, and apparently MDPS expects to hear certain CAN frames on the bus before actually enabling the steering assist - just connecting ignition is not enough. If you have CAN dumps for the vehicle start, could you share those so that I can try to fool the MDPS ECU into working?

Thank you, and really appreciate your sharing this project with the community!
-Anton

k-cleary, 2017-07-03:
Those look like the same part numbers that we have on our cars. It sounds like you may have a different connector, or are possibly measuring different pins.
We don't know much about that model of KIA, so you will need to reverse engineer it yourself. The torque sensor pins will change their voltages when the steering wheel is turned, which should help you identify which pins you need to be reading from.
We've only ever tested the steering on the vehicle, so we don't know that specific CAN sequence. You'll have to find it yourself, which may be a bit tricky.
Thanks for reaching out, and we're excited to see what you can come up with!
-Katie

waiwnf, 2017-07-04:
Great, thanks for the update!

yapjiaqing, 2017-08-15:
Dear PolySync people, what's the difference between the systems of the Kia Niro and the Kia Soul to take note of, that affects the use of the OSCC development kit? We'd like to do all we can on the vehicle first while waiting for Kia Niro support to be out at the end of the year

nathanaschbacher, 2017-09-25:
Howdy,

From our initial testing at the actuation-level the Niro's steering control is different enough to require doing more work determining where to properly tap into the system to enable OSCC.

On the data side of things, the Niro's OBD II port also behaves differently than either the Soul or Soul EV, so there's work for us to do there to make sure we can properly read and parse relevant data from the vehicle.

The soon-to-be-available OSCC hardware updates that support the Soul EV look like they'll also be compatible with the Niro and that the differences will likely be only in firmware updates, but there's still some more work to do for us to be 100% certain.

closed 2017-09-25 by nathanaschbacher

fabio71c, 2019-07-12:
Anyone using comma ai Openpilot / OSCC on a Kia 2015 ?
I have a Kia 2015, I have bought the EON ( Openpilot ), yet since the car didn't come with ACC or KLAS, I'm looking into the OSCC kit.  Would be great to hear if anyone has done such project. Thanks.