PolySync/oscc #156: Need a few more clarifications from the “Overview" Page Diagram

ubgit009, 2017-06-29:
Thank you Dan for answering my first set of questions. Now that we know the diagram in the “Overview” page is the one to be followed, I’m hoping a few more things can be clarified.

1. Where in the Fuse Box, should the wires going to the 200 amp relay and the Emergency Stop be connected? 

2. There are two wire branches coming out of the + terminal in the battery: one is 4 awg and the other 12 awg. Is the diagram implying that we connect two separate lug terminals of different gauges to the battery terminal? 
3. The 200 A fuse, 200 A relay and the System Switch are closer to the + terminal of the battery, than the Fuse Box. Does that mean we need to cut and splice any stock power wires from the Kia Or how do we connect those 3 elements between the battery terminal and the Fuse Box?

4. For any component that connects to ground. Is it better to find a local ground point and try to make a connection there, or to wire it back to the – terminal of the battery? Take as examples the 200 amp  and 40 amp relays, as well as the two pins in the Brake Actuator.

dan-fern, 2017-06-29:
The "fuse box" listed in the "Overview" diagram refers to an added 3rd party fuse block; it does **not** refer to the stock Kia fuse box.  I would suggest finding a fuse block to distribute the +12VDC power from the battery/alternator.  It should serves as a shared grounding point for OSCC hardware, all of which should be then be grounded at the same chassis ground point as the negative battery terminal.  I would also suggest finding a circuit breaker to separate the battery (line) from all the OSCC hardware (load).  I hope that helps.

Thanks for your continued interest in OSCC.

closed 2017-06-29 by dan-fern

ubgit009, 2017-07-04:
Once again, my intention is not to annoy you with a lot of questions. I just want to get past the hardware installation so that I can focus on other aspects of the project. 
I have a few remarks referring to the same diagram:
  1) You mention the fuse box should be used to distribute power, but  in the diagram all I see is a single wire coming in and a single wiring coming out. And there are fuses for all the modules elsewhere in the schematic. So what exactly is supposed to go inside that fuse block?  
   2) It was mentioned in a previous issue that the system, with more hardware in addition to the OSCC modules and gateway used about 85 A. But the 2nd relay (the one that is controlled by the Emergency stop) is rated for 40 A. Is that right? Where is the additional amperage suppose to come from?
  3) If I may make a suggestion. Why not update the "Overview" page to include a list of all suggested parts (along with links to vendors) needed to implement the system on the diagram? I'm sure that will make life easier for a lot of people.