PolySync/oscc #157: About the Control Can Bus Physical Implementation

ubgit009, 2017-06-29:
How do you build the Control CAN Bus that will connect the gateway, OSCC modules and the control computer? As I understand, the bus needs to be terminated at both ends with 120 Ω. What connectors/cables/wires/terminators do we need?

dan-fern, 2017-06-29:
Once you assemble all the OSCC modules, you'll need to connect the CAN Gateway Module to a control computer, or ECU.  This ECU will be used to issue commands to the steering, throttle, and brake modules. 
 

This is a [great overview on CAN networks](http://www.ti.com/lit/an/sloa101b/sloa101b.pdf) from Texas Instruments.  I hope that helps.

Thanks for your continued interest in OSCC.

closed 2017-06-29 by dan-fern

ubgit009, 2017-07-04:
Thanks for sharing that link to Texas Instruments guide on CAN networks. I'm understand how CAN network goes as far as the frame formats, and the overall architecture .  I'm more confused about how to physically connect all the wires. Unfortunately, the guide from Texas Instruments doesn't address that topic in a lot of detail.

Let me be a little more specific:
 1) How do we connect 3 or more nodes  ( i.e. the OSCC 3 modules and the Gateway) to the same CAN wire? Should we do soldered or crimped tap joints? 

 2)Do we need any specific connectors for each node, or bare 20 awg wires would be sufficient? I noticed the DB-9 is popular even though half of its pins are left unused.

 3 According to the  ISO 11898-2 standard, which specifies physical Layer for a  high speed CAN bus, the CAN H and CAN L should be terminated at two **and** **only** two nodes with 120 ohm resistors.
I noticed from the schematics that  the CAN Gateway, the OSCC sensor shields and the brake actuator control board are all terminated with a 120 ohm by default. Do we need to make any changes to the boards, so that only two nodes have the CAN wires terminated? I also read about using separate terminators so that nodes can be removed without electrically compromising the bus.

 4) Do we need a specific cable?  
Appropriate wire gauge? ? Shielded vs Unshielded cable? Characteristic impedance?
 What about the cable shield carrying the reference ground? Please see Figure 1.3.1 and Figure 1.3.2 on page 4 of the following article and you will see what I'm talking about: 
http://canlab.cz/pages/download/CAN-Wiring.pdf