PolySync/oscc #158: Software stack

billtyler, 2017-06-30:
I saw mention in 141 that you use a Alienware Aurora R6 as your compute platform.
What software stack are you using?

k-cleary, 2017-06-30:
Hey, Bill. 
The Alienware ECU is not actually our compute platform for OSCC. It was mentioned as an example of measured power consumption for a vehicle supporting OSCC, sensors, and computer hardware. 
Thanks!
-Katie

closed 2017-06-30 by k-cleary

billtyler, 2017-06-30:
Ok sure, but can you still lets us know what software stack you’re using for driving the car?  It would be nice to get some insight to what is working with the components you have provided while we’re getting our installations under way.  And if not the mentioned pc, what are you guys using for hardware?

k-cleary, 2017-07-03:
Generally, we use ThinkPad laptops with an Intel quad-core 2.20 GHz CPU and Qotom fanless ECU's with Intel Celeron 3215U 1.70 GHz CPU. 
The OSCC software stack uses CMake as our primary build system. Our module firmware is written in C/C++ and utilizes some Arduino libraries, along with an MCP CAN lib for CAN communications and a DAC MCP lib for sending analog signals to the modules.
On the test side, we use Cucumber-CPP for the unit tests, and Rust QuickCheck for the property based tests. 
Joystick Commander uses SDL2 libraries for interpreting game controller commands and sending them to the OSCC modules to be actuated on the car.