PolySync/oscc #165: Basic Electrical Specifications of the OSCC Sensor Interface and Brake Actuator Boards

ubgit009, 2017-07-14:
I realize the Wiki is not intended to be an official source of instructions for installing the kit. Hence, for the purpose of selecting appropriate wire gauges, fuses ratings, and other suitable components to build a power distribution circuit could you please share some of the **basic electrical parameters of the sensor and actuator control boards**?

I'm looking for basic things such as...

-Vin range at which each board can function properly.
- Input resistance of each board.
-  Min/Typical/Max Current used by 
   (1) OSCC Brake Module 
   (2) OSCC Gateway +  OSCC Throttle + OSCC Steeering combined
   when connected to the car modules.
- Any other data you thing would be helpful for the purpose I described above.  

A few side questions on the "Overview Schematic".

I see the circuit is protected with a 200 Amp component next to the battery. Is that supposed to be a fuse or a circuit breaker? Or can it be either?

I did some research on automotive wire ratings by gauge. From what I understood,  there are multiple factors that determine how much current a wire of a given gauge can safely tolerate. However, I found a few articles suggesting that 200 A might be too much for a 4 awg wire. Is 4 awg, as shown in the diagram OK? Alternatively, would lowering the fuse/circuit braker and relay  rating to say about 100 A or 150 A still be acceptable for the application?

Is a 12 awg wire OK to supply power to the OSCC brake module?

Thank you for your continuous support.

LucasBuckland, 2017-07-22:
For both boards:
Vin = 6-26V (60V load dump protection and reverse bias protection)

Steering or throttle module: 60mA average up to  ~100mA.

The brake module pulls 300 mA average up to ~2.5A WITHOUT a brake actuator connected. The electrical characteristics of the Prius actuator can be found online; the actuator itself draws most of the current that goes into the brake module. 

The input resistance would be based on what input you are measuring it at. 

That system diagram is outdated and needs review, I would not rely on it to design a power system. It will be updated in the near future as part of our refresh of the wiki and documentation. 

For your other questions please reference issue #163. In addition this link may be helpful. 

http://www.engineeringtoolbox.com/wire-gauges-d_419.html

ubgit009, 2017-07-23:
Lucas,Thank you for the reply. I'm glad to know the Wiki refresh is on the works.  Our group has been stuck in the installation/ hardware setup phase for months and some of us are eager to make progress. Could you please give us a conservative timeline on when the Wiki Refresh will be completed and available to the public?

http://www.engineeringtoolbox.com/wire-gauges-d_419.html

closed 2017-09-25 by LucasBuckland