PolySync/oscc #178: Does OSCC suport 2017 KIA Niro Hybrid 1.6L DCT in China ？

dongjiahang, 2017-08-22:


nathanaschbacher, 2017-08-22:
Hi,

At the moment the Kia Soul ICE (internal combustion engine) platform is the only vehicle officially supported.

However, coming in the next few weeks the OSCC firmware and sensor interface board designs will be updated with support for the Kia Soul EV. The EV platform won't require the brake actuator installation procedure because we've figured out how to control the factory braking system in the EV, which saves a lot of .

Support for the Kia Niro Hybrid platform is currently on our roadmap to be completed and shipped before the end of the year, but we're always interested in finding development partners to accelerate our timeline for supporting new vehicles. We expect the Niro to be similar to the Soul EV in terms of ease of integration, but provide a driving-range closer to that of the Soul ICE, which is why we're excited to try to support it after we get the EV platform support released.

Soul ICE support will also continue for people who need a versatile long-range platform that is widely available globally today.

Does this also address issue #177 that you created?

dongjiahang, 2017-08-23:
What you mean is that the current open platform does not support Niro？Hardware does not support ？Or software does not support?

nathanaschbacher, 2017-08-23:
In this case both the hardware and the software don't support the Niro. That support is expected to be available before the end of the year (2017). We haven't brought a Niro into the shop to fully characterize the Niro's interfaces, so that we can determine what modifications need to be made to the hardware and/or the software to support it.

Early testing indicates that the Niro will probably be generally similar to the Soul EV, but each platform has its own little differences we need to figure out and integrate into OSCC.

closed 2017-09-25 by nathanaschbacher