PolySync/oscc #201: Steering module not working as expected

ramanathanrm95, 2017-09-22:
I wired up and flashed all the modules without any errors.
I installed the joystickcommander package and followed the steps with no errors.
When I run the ./joystickcommand, the command window initializes and when I press start button, the values of steering, brake and throttle are displayed and when the left stick is toggled, the steering value updates, but after a second, program closes by giving disable controls message and there is no change in the steering wheel position of the car even when the values are updated.

I built the can-gateway module by stacking 2 CAN shields and connecting one of them to Kvaser converter and then to laptop (Modified CS pin of one of the shields to pin 10 --> Connected to Kvaser converter, did not change the resistor)
Connected the steering module as given in GitHub.
Powered all the modules through the laptop itself.
Connected the joystick to the laptop.

Should there be a control can bus separately or can I just use the laptop as control can bus?

This is my current connection:

STEERING --> CAN GATEWAY --> OBD CAN
LAPTOP (CONTROL CAN) --> KVASER --> CAN GATEWAY
ALL MODULES POWERED THROUGH ARDUINO connected to LAPTOP
I don't have a physical control can bus. I am assuming my laptop will function as a control can bus.
No non-e power source.

Please let me know where I am wrong.

austinjmorlan, 2017-09-25:
All of the modules must be connected to the Control CAN bus (your computer will not act as a Control CAN bus), meaning that a twisted wire pair must connect from each module's CAN terminal to the next terminal in a daisy chain fashion.

So one of the CAN Gateway's CAN shields must be connected to the car's OBD bus. The second shield (the Control CAN shield) must then have a twisted wire pair running to the CAN terminal of the first of the modules. Then from that terminal another twisted wire pair must run to the second module's CAN terminal. And again from that terminal to the third module's CAN terminal. That module's CAN terminal should then have another twisted pair running back to the CAN Gateway to complete the circuit.

From the Control CAN shield of the gateway you can then run a final twisted wire pair terminated with a 120ohm resistor and a DB9 connector that you can connect your laptop to.

bobjohnson2040, 2017-10-19:
@austinjmorlan Why did you mention that the first and third module should BOTH be connected to the CAN gateway "to complete the circuit"? The CAN wiring should be a line, not a loop. 

CAN Gateway --> Module 1 --> Module 2 --> Module 3 --> 120ohm terminated DB9 connector

austinjmorlan, 2017-10-19:
@bobjohnson2040 

That's correct, I meant to say that the final module should end in a terminated DB9 connector.

austinjmorlan, 2017-10-30:
@ramanathanrm95 Did my response resolve your issue?

austinjmorlan, 2017-10-31:
@ramanathanrm95 You could attempt to connect to a single module directly (e.g., throttle) from your laptop and see if you're able to communicate with just that one. If so, then there is something wrong with your wiring when you connect multiple.

Run the DB9 connector from one of the module's CAN terminals and connect your Kvaser with a 120ohm terminator to it. Then run joystick commander and attempt to enable.

ramanathanrm95, 2017-10-31:
Will try your suggestions and let you know.
Thanks for the advice. I really appreciate it.

closed 2018-03-12 by nathanaschbacher