PolySync/oscc #205: Brake pump motor doesn't switch off, overheat

rg4444, 2017-10-01:
Using actuator board 1.2.0 after initialization and pressure creation pump motor doesn't switch off.
Here is the test video: https://www.youtube.com/watch?v=tR9DBHNbcTs No errors seen on serial monitoring port. All property tests passed.

LucasBuckland, 2017-10-04:
Hello, this sounds like you have over-pressured your accumulator. This will cause the pump to keep pumping as the pressure reading is incorrect. Please see issue #131. Which pump power connection are you using? To release the pressure in your accumulator you will need to manually actuate the release valves without the pump pumping.

rg4444, 2017-10-04:
We use 2 wire sets. In one we use both, in one only BM1. Symptoms are the same.
According this Toyota diagram BM1 and BM2 is simply hard connected inside module.
[image]
How "manually actuate without pump" should be done?
How it is possible to check the pressure readings from pump?

LucasBuckland, 2017-10-19:
BM1 and BM2 are definitely separate from a functional standpoint. Using BM1 significantly increases the longevity of the motor. I'm not sure where that diagram came from but it's different from what I've seen. 
There are a couple utilities that are useful; included in the Brake firmware folder. One is the "release_pressure" utility which should be useful in your situation. I highly recommend caution when using any of the utilities; having a full knowledge of the system and being careful to re-flash normal firmware when you are done is essential.

austinjmorlan, 2017-10-30:
@rg4444, did @LucasBuckland resolve your issue?

rg4444, 2017-10-30:
We haven't try yet. Will do after a week and put an update

rg4444, 2017-11-10:
We are using BM1 (pink) wire. The problem isn't stable. One day pump is switching off, one not with the same joystick driver and firmware. Cause is still unclear.

rebpdx, 2018-01-22:
@rg4444 Is the oscc git describe you have compiled for joystick commander with the same as the firmware you flashed on to the brake board? When the pump is staying on does it reach and maintain a pressure?

rg4444, 2018-03-09:
I have narrowed down the cause. Is seams there is pressure inside the system. 
The braking system get back to normal operations when pressure lowering after system standby or Utility firmware "release-brake-pressure" is used.
If you have wrote special pressure release utility - there was a reason for it.
Could you elaborate?
What causing pressure problem and how to resolve it without reflashing Arduino time to time?

rg4444, 2018-03-09:
Yes - we are testing and have built latest joystick software together with latest OSCC

rg4444, 2018-03-09:
Started new thread on https://github.com/PolySync/oscc/issues/265

closed 2018-03-09 by rg4444