PolySync/oscc #206: Throttle control problem - no full acceleration

rg4444, 2017-10-01:
On Kia Soul diesel version acceleration control while standing still is working correctly - joystick change value from 0 to 1 and rpm reaches maximum, however while in drive acceleration is week even value is 1. Rpms are not exceeding 2000. Any workaround?

k-cleary, 2017-10-16:
Hey, @rg4444 
Which version of the firmware are you using? If you have upgraded to v1.0.0 or greater, you will also need to update your version of joystick commander. 
The throttle spoofing should work the same whether the car is in park or in drive. 
You may want to check if [stability control](http://www.kisouman.com/electronic_stability_control_esc_-669.html) is enabled. If it is enabled, it might be limiting acceleration.

austinjmorlan, 2017-10-30:
Did @k-cleary resolve your issue?

rg4444, 2017-11-07:
We got the latest version of boards, firmware and joystick commander. 
The voltage
Disabled: 0.67.  3.79  Enabled:   0.79  1.37
Disabled: 0.33.  1.79  Enabled:  0.39   0.68
The stability control is disabled
Problem remains

austinjmorlan, 2017-11-09:
Are there any warning lights no the dash?

rg4444, 2017-11-09:
If you mean car's diagnostic error lights - no, there are no any malfunction indicator lights. And there is no reason for that - when OSCC in enabled - voltage values are in expected range - simply performing only as half way pressed accelerator pedal.

austinjmorlan, 2017-11-09:
Am I correct in understanding that you're using the diesel version of the Kia Soul? We haven't tested the diesel version here, so it's possible that the voltage ranges we supply are not valid for your car.

With the car on but not running, and OSCC powered but disabled, you can take a digital multimeter and measure the voltages on the throttle board's "SIG IN A" and "SIG IN B" terminal connections. Measure both with the accelerator pedal not depressed at all, make note of the voltages, and then measure them again with the throttle pedal fully depressed. Compare those voltages to the [ones we've supplied](https://github.com/PolySync/oscc/blob/v1.1.1/api/include/vehicles/kia_soul_petrol.h) (`STEERING_SPOOF_LOW_SIGNAL_VOLTAGE_MIN`,`STEERING_SPOOF_LOW_SIGNAL_VOLTAGE_MAX`,`STEERING_SPOOF_HIGH_SIGNAL_VOLTAGE_MIN`,`STEERING_SPOOF_HIGH_SIGNAL_VOLTAGE_MAX`

rg4444, 2017-11-09:
Yes, it is diesel version. Will make measurements tomorrow and write an update.

rg4444, 2017-11-14:
With the car on but not running, and OSCC powered but disabled, voltages on the throttle board's are:
"SIG IN A" 0.56 to 3.67 V (default 0.7 - 4.1 V)
"SIG IN B" 0.23 to 1.70 V  (default 0.3 - 2 V)
But output is still almost half less.

cotos, 2018-01-22:
Hi @rg4444,

Please run the following test

* Put car in park
* start car
* start JoystickCommander
* `candump -l can0`
* throttle to max
* throttle to min
* kill candump

repeat test with car in drive.

Attach logfiles

rg4444, 2018-03-09:
This is resolved by compiling/updating throttle firmware. By some reason firmware compiled from source before August 2017 got the wrong voltage - only half of required value.

closed 2018-03-09 by rg4444