PolySync/oscc #212: How to enable only 1 or 2 OSCC modules?

bobjohnson2040, 2017-10-19:
How do I enable only 1 OSCC module? For example, If I want a human to control the gas/brake but I want a computer to control the steering then how would I do this? thanks

bobjohnson2040, 2017-10-19:
This is needed for porting OSCC to other vehicles besides KIA and for testing/debugging.

austinjmorlan, 2017-10-19:
Move the following lines from `api/src/internal/oscc.h` to `api/include/oscc.h`:

```c
oscc_result_t oscc_enable_brakes(
    void );

oscc_result_t oscc_enable_steering(
    void );

oscc_result_t oscc_enable_throttle(
    void );

oscc_result_t oscc_disable_brakes(
    void );

oscc_result_t oscc_disable_steering(
    void );

oscc_result_t oscc_disable_throttle(
    void );
```

Then you can call those functions directly when you include `oscc.h`.

austinjmorlan, 2017-10-30:
Did that answer your question?

closed 2017-10-31 by austinjmorlan