PolySync/oscc #227: Board Files BOM

ahawcker, 2017-12-11:
In the bill of materials of the boards the column "populate" is empty for some parts. I think that all parts which are not labeled with "dnp" should be populate but i didn't know exactly. Can anybody help?

snewt, 2018-01-22:
Hey there @ahawcker , are you exporting the BOM from a `.sch` file with eagle? I'd like to be on the same page of the same document.

ahawcker, 2018-01-22:
Hi, exactly i'm exporting the BOM from the .sch file with eagle. For example the sensor_interface_V1_1_0.sch, what is with the pin strips (Arduino JP) or the connection terminals (Phoenix_4_45)? I think the most of the parts which have no entry at the populate column should populate. Am I right?

snewt, 2018-01-23:
@ahawcker the BOM associated with each board was not intended to be exported from the schematic so there's no guarantee that what eagle produces from the `.sch` files will be sensical. We recognize that documentation is a valuable resource  and are working on making something available.

snewt, 2018-02-06:
Hey @ahawcker , a BOM for each board was added to the `devel` branch. For example, the BOM for gateway lives here `https://github.com/PolySync/oscc/tree/devel/hardware/boards/gateway`. They are comma delimited CSV files that can be imported into spreadsheet software like Excel or LibreOffice Calc.

commit 2715b284 (2018-02-06): Add BOM for each board

Prior to this commit there was not a reliable BOM associated with the
boards we tracked schematics. This commit represents the addition of
the BOM for each board as well as a corresponding MacroFab XYRS data
file. Each of these files are in a comma delimited CSV format that
can be imported into spreadsheet software.

Closes #227
closed 2018-02-06 by snewt