PolySync/oscc #263: Definition of Pins in \brake\include\globals.h

rg4444, 2018-03-02:
There are configuration of Pins like " #define PIN_ACCUMULATOR_PUMP_MOTOR_CHECK ( 8 )"
What is reference for pin numbers? 
What could be reason to change the values of pins?

adrienthebo, 2018-03-02:
The pin definitions are based on the actuator control module (https://github.com/PolySync/oscc/blob/master/hardware/boards/actuator_control/actuator_control_1.2.0.pdf). The only motivation I can think of for changing these pin definitions is if you're designing your own brake actuator PCB.

adrienthebo, 2018-03-02:
Since this issue appears to have been addressed I've closed this issue, if you have additional questions please reopen this issue and we can continue this discussion.

closed 2018-03-02 by adrienthebo