PolySync/oscc #264: AVRDUDE read and write firmware to backup and manage firmware versions on Arduinos

rg4444, 2018-03-08:
There is handy tool call `avrdude` to read and write firmware to Arduino's.
Here are example codes:
Read:
`sudo avrdude -patmega328p -carduino -P/dev/ttyACM0 -b115200 -Uflash:r:/OSCC_Firmware/Read/firmware-throttle.hex:i`
`sudo avrdude -patmega2560 -cstk500v2 -P/dev/ttyACM0 -b115200 -Uflash:r:/OSCC_Firmware/Read/firmware-braking.hex:i`
Write:
`sudo avrdude -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -V -Uflash:w:/OSCC_Firmware/Write/throttle.hex:i -Ueeprom:w:/OSCC_Firmwares/Write/throttle.eep:i`
However the size of .hex file sent to controller is smaller than downloaded.
What could be cause? Or what are the correct read and write commands to maintain correct structure of firmware in hex file?

shnewto, 2018-03-09:
Hi @rg4444 
We rely on some default configuration to upload firmware to the Arduinos using avrdude. That write command for our system is a bit different from the one you've referenced.
```bash
avrdude \
-C/usr/share/arduino/hardware/tools/avrdude.conf \
-patmega328p \
-carduino \
-b115200 \
-P/dev/ttyACM0 \
-D \
-V \
-Uflash:w:OSCC_firmware/Write/throttle.hex \
-Ueeprom:w:OSCC_firmware/Write/throttle.eep:i
```

rg4444, 2018-03-09:
Thank you! 
But how about read it back?

PML-Trey, 2018-03-09:
Change the :w: to :r: and set the filename you want it to write to.

Trey

rg4444, 2018-03-09:
I did it.
But I concerned about different size of hex files.
If uploaded files are 52,8 and 52,4 kB for throttle/steering, then downloaded is constantly 77,8 kB.

shnewto, 2018-03-09:
Hi @rg4444 ,
I don't have a ready answer about the size discrepancy you're reporting. I may be able to investigate further if you can share the output of the command `avr-size file.hex` for both the uploaded and downloaded files.

rg4444, 2018-03-09:
[read_write_examples.tar.gz](https://github.com/PolySync/oscc/files/1798699/read_write_examples.tar.gz)
Here are 3 different firmware writes and reads.
write
`sudo avrdude -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -V -Uflash:w:/OSCC_Firmwares/public_examples/can-gateway.hex -Ueeprom:w:/OSCC_Firmwares/public_examples/can-gateway.eep:i -C/usr/share/arduino/hardware/tools/avrdude.conf`

read
`sudo avrdude -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -V -Uflash:r:/OSCC_Firmwares/public_examples/can-gateway_read.hex:i -C/usr/share/arduino/hardware/tools/avrdude.conf`
please mind missing :i in read command. Without it script through out error

PML-Trey, 2018-03-09:
The firmware images you're programming will only contain the memory needed
for the image while reading flash will read the entire flash.  This seems
consistent with the file sizes you're seeing.

closed 2018-03-10 by shnewto