PolySync/oscc #265: Brake pressure release problem - overpressuring - on KIA Soul Petrol

rg4444, 2018-03-09:
After continuously (more than 10 min) excessive braking, ABS unit's  internal brake pressure is increasing, pump motor doesn't switch off, overheat.
Only solution until now is to run firmware : https://github.com/PolySync/oscc/blob/master/firmware/brake/kia_soul_petrol/utils/release_pressure/src/release_pressure.ino 
What is the cause for this behavior? Any fixes?

adrienthebo, 2018-03-09:
Hmm, this seems like a worrying issue. I'm not sure that I fully understand everything that's going on, especially since we haven't encountered issues with the ABS unit overheating during testing. In order to help you out I'll need some clarification about some of the details.

  1. What do you define as excessive braking? What brake position were you commanding during this testing? Is the brake pressure applied outside of the range of normal pressures that the vehicle would encounter during an emergency situation?
  2. Was the vehicle in motion and you were testing braking, or were you continuously applying braking as a stress test?
  3. Can you provide the code that caused this behavior?
  4. You mention the ABS unit - are you referring to the electric brake actuator that's installed in series with the standard Kia Soul brake actuator, or is there another stock vehicle system that's overheating?

In addition we have a standard template for filing issues with fields, filling that out would help us get to the bottom of this issue more quickly. Could you fill it out?

### Expected behavior
What should be happening?

### Actual behavior
What behavior are you seeing?

### Steps to reproduce
How can we make it happen too?

### Version info
-  What's the output of running `git describe --tags` in your OSCC directory?
- Is this the same version flashed onto the hardware?

### Hardware info
Are you using a custom board or something received from us?

rg4444, 2018-03-09:
See comments below.
But most important is answer to question for need of special utility for pressure relief.

What do you define as excessive braking?   - Accelerate up to 15mph then stop. More than 20 times in 2 minutes.
What brake position were you commanding during this testing?  -- Joystick control. 
Is the brake pressure applied outside of the range of normal pressures that the vehicle would encounter during an emergency situation? --Not sure. Mild braking at low speeds.
Was the vehicle in motion and you were testing braking, or were you continuously applying braking as a stress test? --Vehicle is in the motion.
Can you provide the code that caused this behavior? -- Latest version of firmware, compiled for KIA Petrol + debug flag.
You mention the ABS unit - are you referring to the electric brake actuator that's installed in series with the standard Kia Soul brake actuator, or is there another stock vehicle system that's overheating? --I am referring to Toyota ABS module.

Expected behavior.  -- Normal braking behavior
What should be happening?
Actual behavior
What behavior are you seeing? --
Steps to reproduce  -- Run vehicle for 10 minutes start/stop cycle - to accelerate up to 20 mph and brake by joystick control.
How can we make it happen too? --
Version info
What's the output of running git describe --tags in your OSCC directory?  -- Should be latest version cloned March 8 , 2018
Is this the same version flashed onto the hardware?
Hardware info
Are you using a custom board or something received from us? --Braking board made by/delivered by Polysync August 2017

shnewto, 2018-03-09:
Hi there @rg4444 
I'm in favor of getting as much information as possible about this issue so I'll also vote for some of the above template filled out. 

As far as potential fixes go however, I can recommend that you try using our brake-monitor. Here's how I suggest it's used.

1. Install the `screen` application. `sudo apt install screen`.
1. Assuming you've already built the firmware and have an `oscc/firmware/build` directory, navigate there. (otherwise, from the `oscc/firmware` directory, run `mkdir build && cd build`)
1. Build the project with debugging on then run the brake-monitor.
```bash
cmake .. -DYOUR_TARET_PLATFORM=ON -DDEBUG=ON ..
make brake-monitor
```
1.  That should launch the brake-monitor. Note that from here, you can run `<ctrl> \` to exit. 
1. The following sequence of keys is  `m` `j` `k` `,` is supposed to restore a normal state.
1. If it doesn't, you may have to toggle the master cylinder solenoid with the `i` and `u` keys (until the pedal does not exhibit any "firm resistance".)
1. If step 6 is required, use `m` `j` `k` `,` to get back to a normal state once again before exiting. 

For some context, the `m` key is to engage the solenoid, `j` is the command to release pressure in the accumulator, `k` and `,` are to close the actuator and release.

rg4444, 2018-03-09:
Hmm.. Sounds like known problem.
Will try.
Is there a chance to fix it in production environment?

shnewto, 2018-03-09:
By production environment do you mean without the debug flag enabled? If so, the debug flag is only required to use the `brake-monitor` application to restore the brakes to a good state. After that, feel free to restore the firmware to a version without debug flags. 

Also, I forgot a couple steps above. The firmware built with the `-DDEBUG=ON` needs to be uploaded to the brake module and when running the `brake-monitor` application you need to be connected to the brake module through USB.

rg4444, 2018-03-09:
I meant production environment when end user could run system flawlessly. Without need to restart, reboot, reconfigure etc

rg4444, 2018-03-20:
I've tried all keyboard buttons in 'screen' terminal window you referred, but seems non of them work. Any pre-requisites for building, launching or pressing keyboard?

shnewto, 2018-03-20:
@rg4444 it looks like I did leave out a step!
While connected to the brake module via USB run the following commands. 

```bash
cmake .. -DYOUR_TARET_PLATFORM=ON -DDEBUG=ON ..
make brake-utils-serial-actuator-upload
make brake-monitor
```
Next try the keyboard sequences again.

rg4444, 2018-03-24:
Thank you! Now its working. All solenoids are operating by key strokes. Biggest problem is that pressure debug readings are not implemented on joystick controllable firmware version.