PolySync/oscc #266: throttle fault issue

billtyler, 2018-03-12:
I have a few feet of wire for the throttle sensor run up to the passenger seat for when I hook up the computer for testing etc.  This has worked without issue for a few months, however in the last week or so it has faulted twice.  I'm currently sitting at a Starbucks waiting for it to reset and allow normal driving again.

Based on the few times I have experienced this I know it will reset after some time, does anyone know how long it takes for this?

Any ideas on making sure it does not trip for simply jostling the wires?

shnewto, 2018-03-12:
Hey there @billtyler ,
I'd recommend picking up a diagnostic tool like the Launch Creader if you're able (https://www.amazon.com/Launch-Creader-Scanner-Functions-Diagnostic/dp/B01N36W561). That should provide more specific context around your faults and hopefully enough information to help you spot the problem that's causing it.

closed 2018-03-12 by shnewto

billtyler, 2018-03-13:
I was able to get a get a scanner and reset two of the three codes.  It drives now but there is a code listed as permanent - P2127 Throttle/Pedal Position Sensor / Switch "E" Circuit Low Input.  Are "permanent" code only able to be reset by the Kia device?

rebpdx, 2018-03-15:
I'm not sure about permanent codes and which devices do and do not reset them. I'd recommend making sure there aren't any intermittent connections or any high impedance for the Pedal Position Sensor described by the fault and then try resetting it.