PolySync/oscc #268: Unequal brake pressure FR and FL wheels on KIA Soul Petrol Toyota ABS

rg4444, 2018-03-22:
We are experiencing unequal brake pressure using joystick commander.
Here is test video: [https://www.youtube.com/watch?v=1pKUC6Q5qEs]
In the same test environment manual brakes are working equally on all 4 wheels.
After entering joystick mode:
- rear wheels are braking equally;
- front wheels in low-to-middle braking forces are working unequally. Difference is more than 50% which is not permitted by regulation. In our case more pressure is applied to front left wheel.
Is there a way how to adjust pressure in firmware settings, as well as monitor software pressure value output in real time?
What was a reason to not include pressure monitor values in debug version of brake board firmware?

cotos, 2018-03-22:
Hi rg4444,

This seems like a mechanical issue to me.
* Verify there is no air in the brake lines going to the front right and left brakes.
* Uneven brake pressure may indicate an issue with the Toyota ABS.

The brake system digram is on this [page](https://github.com/PolySync/oscc/wiki/Hardware-Brake-%28Petrol%29)

We do not have firmware settings to adjust brake pressure for individual brake solenoids.
We do have a "Serial Actuator" utility that will monitor brake pressure on the front right and left brake solenoids.

### Make the “Serial Actuator” utility.
```
$ cd ~/OSCC/firmware && mkdir build && cd build
$ cmake .. -DKIA_SOUL=ON -DDEBUG=ON
$ make
```

### Upload the Serial Actuator utility
Connect the actuator module to a laptop to using a USB A-B cable. 
`$ make brake-utils-serial-actuator-upload`
 
### Monitor the serial output over USB.  
`$ make brake-monitor`

Note:  screen must be installed 
`$ sudo apt install screen`

### Exit Serial Actuator utility
```
$ <ctrl> a \
$ yes
```

### Measurements we see when running this test:
Test set up for brake-monitor test
Car is off
Power is applied to brake module (if you hear the brake pump during the brake pump verification step power is applied)
Verify you can hear the solenoids activate and deactivate
Press "m" (you should hear the relay engage)
Press "," (you should hear the relay disengage)
Press "j" (you should hear the relay engage)
Press "k" (you should hear the relay disengage)
Test the master cylinder
press the "i" key on the keyboard to close master cylinder solenoid
press the brake pedal
**CAN,0,PACC,12.00,PFL,12.00,PFR,12.00,PMC1,172.70,PMC2,199.10**
press the "u’’ key on the keyboard to close master cylinder solenoid
press the brake pedal
**CAN,0,PACC,12.00,PFL,153.50,PFR,151.10,PMC1,184.70,PMC2,184.70**
Get to a known state Release all the pressure in the accumulator Press “m” “j” Press “k” and “,” to close the actuator and release solenoids You should see something similar to this **CAN,0,PACC,12.00,PFL,12.00,PFR,12.00,PMC1,12.00,PMC2,12.00**
Press the brake You should see 
**CAN,0,PACC,12.00,PFL,98.30,PFR,83.90,PMC1,100.70,PMC2,117.50**
Verify brake pump is adding pressure and PACC stays pressurized
Press the ‘p’ key on the keyboard. The accumulator, pumping up pressure, should be audible. The PACC pressure should increase.
After 5 seconds press '['
PACC should be ~800
Wait 20 seconds the value should not decrease by more than 2%
You should see
**CAN,0,PACC,842.30,PFL,12.00,PFR,12.00,PMC1,12.00,PMC2,12.00**
Ensure that the master cylinder solenoid is working properly
press the "i" key on the keyboard to close master cylinder solenoid
press the brake pedal
**CAN,0,PACC,12.00,PFL,12.00,PFR,12.00,PMC1,172.70,PMC2,199.10**
test the proper operation of the “actuator” and “release” solenoids to the wheels.
press "," to ensure that the “release” solenoid is closed
press the ‘j’ key to open the actuator solenoids. (You should hear the solenoid disengage)
You should see
**CAN,0,PACC,866.30,PFL,878.30,PFR,878.30,PMC1,12.00,PMC2,12.00**
Get back to a known state
Press “m” (you should hear the solenoid engage, PACC will go to ~12.00)
PRESS “j” to release all the pressure in the accumulator
Press “k” and “,” to close the actuator and release solenoids
You should see something similar to this
**CAN,0,PACC,12.00,PFL,12.00,PFR,12.00,PMC1,12.00,PMC2,12.00**

rg4444, 2018-03-24:
Great tutorial! Thank you
All solenoids are working as expected - pressure  readings are legit - see chart below 
[image]

rg4444, 2018-03-24:
Is there a way how to get the same pressure readings in ordinary brake firmware? What was a reason to not integrate debug pressure readings in base firmware, so pressure could be monitored in fully operational mode , i.e. joystick controller?
Are there any tools to visualize realtime data from _screen_ - to draw realtime charts?

rg4444, 2018-03-25:
Command descriptions:
'u': Master_cylinder_open - "opened SMCs"
'i': Master_cylinder_close - "closed SMCs"
'j': Set_accumulator_solenoid_duty_cycle( SOLENOID_PWM_ON ) -"opened SLAs"
'k': Set_accumulator_solenoid_duty_cycle( SOLENOID_PWM_OFF ) - "closed SLAs"
'm': Set_release_solenoid_duty_cycle( SOLENOID_PWM_ON ) - "opened SLRs"
',': Set_release_solenoid_duty_cycle( SOLENOID_PWM_OFF ) - "closed SLRs"
'p': Accumulator_turn_pump_on
'[': Accumulator_turn_pump_off
https://raw.githubusercontent.com/wiki/PolySync/oscc/images/brake_module/brake_stock_diagram.png

rg4444, 2018-04-01:
After adding debug code to firmware here are the readings from both: joystick control and manual pedal.
[sdp1.xlsx](https://github.com/PolySync/oscc/files/1866624/sdp1.xlsx)
There is difference between readings from FR and FL sensors. For some unknown reason FR pressure are higher than FL. Also FL seemingly lagging in time - by responding later. Its happening even in manual pedal operation. Any clues?

JesseS95, 2019-10-15:
Hi mate,
I'm testing the brake module for the petrol Kia soul.
I find that the relay 3 is flickering.
Should this relay be flickering on and off (sometimes too fast for the pump to even start)  when the brakes are not applied?