PolySync/oscc #289: Male/Female connector part numbers for throttle/brake/steering?

frk2, 2018-05-09:
Hey guys,
Not a bug per-se but more a request for information if someone has done this. I'm looking for part numbers for male/female connectors for the throttle/brake position sensor and the steering torque sensor connector. I'm trying to make clean Y connectors for everything to plug OSCC into my car.

The only thing ive been able to find is that the brake connector housing is this one:
https://www.mouser.com/ProductDetail/571-1-967616-1

Does anyone have more information on what housings they used? Or is everyone splicing wires to get this to work?

Also curious if its possible to pay extra to get all these connectors / pigtails shipped from polysync!

Thank you

DavidSosnow, 2018-05-09:
Hi Faraz,

We've done the hard-work of tracking down or building various connectors, designing automotive-grade wiring harnesses (that tie cleanly into the vehicle), and more in our DriveKit product. Unfortunately, we can't sell just the individual parts, but if you're curious to learn more about DriveKit let us know.

https://polysync.io/drivekit/

frk2, 2018-05-09:
@DavidSosnow I fully understand and respect that. Im actually doing this for a opensource project (https://frk2.github.io/opencaret/) and am unable to pay the full amount ($20k) for drivekit at the moment. I'm willing to pay a heavy premium for these connectors if you guys can simply ship me the harness that you guys use with drivekit! Is that doable?

OAkyildiz, 2019-09-20:
@kenkeiter We understand that, but are you also stating that you do not want to disclose what cable harnesses you are using? 
(For the interested, I found the male connector for KIA Soul EV brake)