PolySync/oscc #299: throttle graph in wiki is missing.

asimay, 2018-07-24:
### Expected behavior

### Actual behavior

**throttle graph in wiki is missing.**

### Steps to reproduce

https://github.com/PolySync/oscc/wiki/Hardware-Throttle

[image]

### Version info
- _What's the output of running `git describe --tags` in your OSCC directory?_
- _Is this the same version flashed onto the hardware?_

### Hardware info

### Proposed feature or changes

### Use case

### Documentation expected

### Documentation available

shnewto, 2018-07-24:
Hi @asimay thanks for pointing this out! I've attached included the image here while we work out updating the wiki.
[image]

shnewto, 2018-07-24:
Image link updated. Thanks again for pointing it out @asimay

closed 2018-07-24 by shnewto