PolySync/oscc #308: Control the Steering Wheel by angle not torque

zhishan-li, 2018-08-19:
@austinmorlan  Is there an API to control the steering wheel by the angle rather than the torque? You known, the torque depends the vehicle speed. BTW, I am workding the KIA_NIRO. Thanks. 

The following is the API given in [ossc.c](https://github.com/PolySync/oscc/blob/master/api/src/oscc.c#L237)

```
oscc_result_t oscc_publish_steering_torque( double torque )
{
    oscc_result_t result = OSCC_ERROR;

    oscc_steering_command_s steering_cmd =
    {
        .magic[0] = ( uint8_t ) OSCC_MAGIC_BYTE_0,
        .magic[1] = ( uint8_t ) OSCC_MAGIC_BYTE_1
    };

    steering_cmd.torque_command = (float) torque;
     global_vehicle_can_socket, global_oscc_can_socket);

    result = oscc_can_write(
        OSCC_STEERING_COMMAND_CAN_ID,
        (void *)&steering_cmd,
        sizeof(steering_cmd));

    return result;
}
```

andrewraharjo, 2018-08-19:
Niro has the LKAS system that you can tap into. You can trigger the steering command at 9 mph. I can’t think of anything that will let you bypass that LKAS/ACC safety. Unless you want to reverse the firmware on Mobis MDPS, that might work out to the upper layer of OSCC.

andrewraharjo, 2018-08-20:
You just need to define your own scale factor. You might want to play with the steering column separately. This action might/might not affect your steering PID tuning. Honestly, I was just throwing you an idea for your problem. If that’s what you want, you might want to loh the steering voltages, time stamp them, and sync it with the OBD steering angle. Once you got those three values, it should be an easy way to create a linear equation/formula for the conversion.