PolySync/oscc #326: Something about PCB

JesseS95, 2019-02-26:
Hi all, 

there are some parts in the BOM but not on the PCB.

sensor_interface : C19

actuator_control: RELAY2 R29 R31 D5

Vehicle_control: R24 R25

What are they used for ?

PML-Trey, 2019-02-26:
sensor_interface : C19
C19 Looks to be an error in the BOM

actuator_control: RELAY2 R29 R31 D5
RELAY2 is the actual relay.  It is socketed and fits in RELAY3.
R29 and R31 are used to seperate the sense nets so they can be named to enable differential routing in EAGLE.
D5 is a diode to protect against back EMF from the solenoid.

Vehicle_control: R24 R25
R24 R25 Look to be an error in the BOM

I suspect the errors are due to earlier versions of the boards having these parts.

closed 2019-02-26 by PML-Trey

JesseS95, 2019-02-28:
Hi Trey, In terms of actuator_control: Where are the position of R29, R31, D5 on the board?

PML-Trey, 2019-02-28:
R29 and R31 are just to the left of CON5 and CON7. D5 is directly behind CON5.